Acill/A4000RevB #1: A4000RevB schematic feedback

mrand, 2018-06-24:
I saw the request for review and had a few minutes to fly through the schematic.  I didn't review much of the logic, so hopefully someone else is doing that.

I haven't looked at the 74F08 and HC logic datasheets in a while,  but I'd suggest tying both inputs off, not just one input of each gate (U130-A, -C, and -D).  U177.13 and .14 as well as U215-D and U907-A.   And maybe the inputs to U459-B.

And when you do, you might consider tying off all the logic gates with pull resistors so that you can more easily use that gate in case of emergency rather than dead-bugging a gate somewhere, or having to lift a pin. 

Is there a way to do a cross-reference (typically used to add page numbers to off page flags)?  That would help you verify that you don't have any nets mis-named or mis-connected.

There are a few comments, including most especially on page 13, that make it seem like some follow-up work might be needed.

Nice work!

Acill, 2018-06-25:
Thanks for the comments! I have submitted this to the company doing the design work.

Acill, 2024-05-21:
Schematics are in what I feel are the final and most closely representing the final production version of the A4000. Many improvements have been done as well to clean up some silly decisions made by the original design.

closed 2024-05-21 by Acill