Acill/A4000RevB #2: ADV7120 does not get 5V VAA

ishe, 2019-01-22:
I have built the A4000 with this motherboard and have noticed a serious issue. The picture on the monitor was too dark when I used the RGB signals (indeed I was lucky to see anything). It turned out that the ADV7120 does not get 5v to its VAA pins. Troubleshooting it I have noticed that L500 inductor is connected to nowhere. It seems the layout is incorrect and ADV7120 does not get 5v input! L500 should be connected to 5v and to AVID+. This is missing from the PCB so L500 is isolated and useless.

Acill, 2019-01-23:
> If we install L500 to the place of R465 or we install an 1ohm resistor to
> R465 then ADV7120 gets the required 5v and works.

Where did you get your board from? I have built several of these now as
well as several other people without finding this issue.

Acill, 2019-01-23:
Where did you get the board you used from. Was it one of mine? Several people have built them now as well as the 4 I have done myself without finding this issue. All are working just fine.

ishe, 2019-01-23:
Hi,
I have bought it from you via ebay. But I am 100% sure all your boards
don't have the L500 connected to anywhere. I have checked the schematic on
Github and L500 is left unconnected in the schematic! The reason why others
did not have issue is that they have installed R465 (1 ohm). I however did
not install it at first because original boards did not have it either.
Also amigapcb.org has L500 connected to null signals. It is weird why.
Istvan

ishe, 2019-01-24:
I have used both locators, amigapcb.org and John Hertell's locator to build. I don't have original board to verify, but from the photos on the net what I can see is that L500 has 2 vias going to probably to vcc layer and to other side. On the photos (big book of Amiga HW) R465 is not installed. Also the BOM on github says for R465 that it is a blank component (number of 1ohm 1206 resistors is only 2). Using this motherboard L500 is useless and should be removed from BOM.
In some other forums at least one person mentions that he has 10k resistor on R465 which obviously not a good choice if there is no L500. Other person has measured voltages on L500 and it was 4.95v so I am sure on some boards L500 should be active.

Acill, 2019-01-24:
I will look more into this then.

Acill, 2019-01-25:
Looking more into this  If you're going to fit a 1 ohm resistor at R465, then dont bother installing L500. With L500 installed you shouldnt need R465. This is shown on the original Commodore schematic as well and dont understand why they have used both, but being a clone i just followed it.

ishe, 2019-01-27:
Yes, that is true. Probably L500 was planned to filter PSU noise. It works
fine with R465 as 1 ohm. Other parts of the PCB seem to be ok. I have built
it and works fine with an A3660 CPU card!

2019. jan. 25., Pén 19:40 dátummal Acill <notifications@github.com> ezt
írta:

johntayloruk, 2020-06-09:
Hi, I know this is old now, but I’ve just spent the entire day trying to identify why my new build wasn’t working.  It’s only when I caught glimpse of something on screen that I turned the brightness up on my monitor that i saw a very faint display.  I’ve checked all components related to the video output and the come across this thread.

I have put the fixed inductor in L500 and a 10k resistor at R465 as per the BOM.

Are you saying I need to remove the 10k resistor at R465 and replace it with a 1 ohm resistor and it will work??  Do I need to remove the L500 top or is it ok there with the 1 ohm resistor?

ishe, 2020-06-10:
Hi,

There is no point in installing L500 to its place on the motherboard as
this replica does not have that connected at all. If you check traces it is
just a copper island, so it does not play a role.
Instead I recommend to go on the bottom of the motherboard and either
install 1 ohm to R465 or install the inductor to R465 which was supposed to
be installed on L500. I have tried both and it works fine. I have left the
inductor there.
Without these the ADV7120 does not get the 5v supply thus your screen is
very dark.

Br
Istvan

johntayloruk, 2020-06-10:
Just added the 1ohm resistor and it works!! Thank goodness I came across your issue, I haven't seen it mentioned anywhere else and my orig A4000 has nothing at R465.
Very happy now! :)
Thank You!

ishe, 2020-06-10:
I am glad I could help you! Your original A4000 must have L500 in its place
and that is the reason why it works without R465. On that motherboard L500
trace is connected and provides the 5v to ADV7120 via that thus R465 is not
needed there.
Best Regards
Istvan

Acill, 2020-06-18:
Put them 1Ohm in and see if it works.. I think it should. If not remove
L500.

saft6luck, 2024-02-06:
> Put them 1Ohm in and see if it works.. I think it should. If not remove L500.

So, in Jan2019 version, L500 is not connected at all. Only R465 can be used.

Acill, 2024-05-21:
New gerbers are clearly marked not to populate this component and to use a 1Ohm in place in the new silkscreen.

closed 2024-05-21 by Acill