Acill/A4000RevB #5: Gerber problems...

JensRestemeier, 2020-06-20:
I'm not having much luck using the Gerber files.  (a4k-6x_pads_Jan2019.zip)
GERBV shows one layer (ASB) mirrored, most layers offset from each other.
The Gerber viewer in KiCad shows the top silkscreen layer offset from the rest.

I tried uploading it to a few board houses to get a quote, but they all seem to error out. Renaming the layers to the recommended names, for example from OSHPARK (https://docs.oshpark.com/troubleshooting/naming-pattern/ ) and deleting extra files still fails because of a missing board outline.

Obviously some people get boards manufactured, so is there a board house that has success with these files? Do they need to be converted somehow?

Acill, 2020-06-21:
These are in PADS format. I have been using a place in the US to do it, but
any reputable board house should have no problem with these. I have heard
even  https://jlcpcb.com/ has done them.

Acill, 2020-06-21:
Use this viewer:  https://www.pads.com/downloads/pads-pcb-viewer/

johntayloruk, 2020-06-21:
I’ve just had a bunch of A3660 boards made through https://jlcpcb.com/ And has no problems.  Easy to upload the gerbers and get a quote, fast build, delivery and reasonable price too.

JensRestemeier, 2020-06-21:
JLCPCB is the place I usually use for my PCBs, and I was trying to get a whole set of motherboard, daughter board and A3660 board.

The zip file from Chucky uploads fine, displays a preview and gives a quote.

The Gerber zip file from https://github.com/arananet/A4000db/tree/master/version1.2 previews fine in the KiCad Gerber viewer, but doesn't display anything or updates a quote on jlcpcb.

The Gerber zip file from https://github.com/Acill/A4000RevB/blob/master/Gerbers/a4k-6x_pads_Jan2019.zip shows misaligned layers in KiCad and Gerbv, and a mirrored layer in Gerbv. It doesn't display anything or updates the quote on jlcpcb.

I tried PCBWAY, JLCPCB and OSHPARK. OSHPARK gave me a quote for the daughter board after I renamed layers to match their naming pattern, it didn't work with the motherboard Gerbers. I didn't try any other company that requires an email registration before doing anything.

If people had success with JLCPCB in the past I'll contact their support, in case a recent software update on their side broke their import pipeline.

closed 2020-06-21 by JensRestemeier

JensRestemeier, 2020-06-22:
I heard back from JLCPCB support, they think the upload problems are because of the file size, and they may work OK.
They pointed out that a4k-6x_pads_Jan2019.zip has no independent outline layer, is it possible to generate that in Pads?
(It makes me a bit nervous to order large boards without checking that the files uploaded correctly into the system, but if other people had success I'm willing to give it a try...)

kavanoz64, 2024-05-15:
I uploaded new Gerber and pick and place files. These work fine with JLCPCB and should work with other PCB houses too.