Acill/A4000RevB #6: Gerber files

AmigaPapst, 2020-12-14:
hello,

we would like to use the gerber files. But they does't work in Kicat. Can you help?

Best regards
Simon

Acill, 2020-12-14:
These gerbers are made using Mentor Graphics PADS. They work find for
manufacturing boards as long as you mention that to the board house you
use. I have had many made using them. I don't have knowledge of other CAD
software and use PADs.

-Paul

AmigaPapst, 2020-12-14:
Thanks for your fast reply. I have problems to view the gerber files with gerber viewer software. We see a offset of different layer. Which files are need to produce the boards?

AmigaPapst, 2020-12-14:
The top silk screen .SST has a offset of x and y coordinates compare to copper layers. The bottom silk screen .ssb fits perfectly.

AmigaPapst, 2020-12-14:
drill file (*.drd) has a y offset of 2,54 mm.

AmigaPapst, 2020-12-14:
The following layer are fits perfectly aligned on top:
[image]

AmigaPapst, 2020-12-14:
The following layer are doesn't fits perfectly aligned on top:
[image]

AmigaPapst, 2020-12-14:
They have the same problems: https://community.sw.siemens.com/s/question/0D54O00006eo69e/output-gerber-files-are-not-aligned

AmigaPapst, 2020-12-14:
This is a example manual how to export gerber files from PADS.
https://docs.oshpark.com/design-tools/mentor-graphics-pads/

Acill, 2020-12-17:
a4k-6x_pads_Jan2019.zip
<https://github.com/Acill/A4000RevB/blob/master/Gerbers/a4k-6x_pads_Jan2019.zip>
is the complete gerber you should be sending to a board house to get boards
made and tell them its in mentor Graphics PADS format.  These are from
professional board cad software, not open source like Kicad and cant be
read by them. If you want to view them you can use the PADs viewer for
free. Its located here: https://www.pads.com/downloads/pads-pcb-viewer/

You can open the project layout files for complete viewing as well. That
file is : a4k-4x_pads_FINAL.pcb
<https://github.com/Acill/A4000RevB/blob/master/Layout/a4k-4x_pads_FINAL.pcb>

Here are direct links from my github project page

Project layout file: https://github.com/Acill/A4000RevB/tree/master/Layout
Gerbers: https://github.com/Acill/A4000RevB/tree/master/Gerbers

Acill, 2020-12-17:
I have produced hundreds of these boards with no issues using the export
file I provided you. If you are using one from another source I cant help
you. Others have also been able to produce working boards suck as john
hertell in Sweden. Whoever you are using to make them is clearly doing
something wrong if they cant figure it out.

Acill, 2020-12-17:
Seems fine here?

[image: image.png]

AmigaPapst, 2021-01-17:
Thanks for your help. We produce the board. 👍

The-Real-G-Evil, 2022-08-06:
Hello Acill

Where did you produce your boards?

I want JLCPCB to produce them and they ask for the layer names.
Can you tell me what to select?

Regards Tim

Acill, 2022-08-06:
The auto tool won’t work. Upload the complete Berber zip and tell them the
format is mentor pads. John Hertell has used them and has no problems  with
them.

kavanoz64, 2024-05-15:
I uploaded new Gerber and pick and place files. These work fine with JLCPCB and should work with other PCB houses too.

Acill, 2024-05-15:
I just merged it. Thanks!

AmigaPapst, 2024-05-15:
Hello Acill,

thanks for you message and for merging. ;)

Best regards
Simon

Kind Regards
-- 
<sb>
Alinea Computer
Torsten Hees & Simon Neumann GbR
Amiga-Softwareentwicklung, Vertrieb und Online-Shop

Geschwister-Scholl-Str. 26
61273 Wehrheim
Deutschland

Tel.: +49 (0) (60 81) 5 76 99 88 
(Zeiten: Mo bis Fr von 18 bis 19 Uhr, Sa von 11 bis 12 Uhr)

Internet-Seite: http://www.alinea-computer.de
Online-Shop: http://www.amiga-shop.net
Facebook: http://www.facebook.com/alineacomputer

closed 2024-05-21 by Acill