Acill/A4000RevB #7: PCB Specifications?

shred, 2021-02-18:
I'm not an expert about PCB production. I tried to calculate the price at pcbway, but after uploading the Gerber files, I was confronted with questions about PCB specifications, like the number of layers, the individual layer names, thickness, min hole size, surface finish, via process, etc.

Can you give the "musts" and also some recommendations in the README that would give a good result?

Thank you!

Acill, 2021-02-18:
Its a 4 layer board and you need to email them the the zip file and tell
them its gerber output fron Mentor Graphics PADs.

-Paul

shred, 2021-02-18:
Even if I would send the zip file by mail, I guess they still want to know about thickness, surface finishing etc. What I mean is: It would be easier for noobs like me to produce "cheap" replica PCBs if I can just upload the ZIP file in their quick order tool, pick some options like recommended in the README, and then place the order and be done with it.

Well, I tried their Gerber viewer and ran into issue #6, so it probably won't work with pcbway anyway.

Acill, 2021-03-03:
Many people have had these made at those places with no problems. Its all
standard specs. Finish is up to you, most recommend Enig. If you use a PCB
viewer use the free one from Mentor PADS.

Acill, 2021-03-03:
Here is an order from JLPCB I made to see how they worked out. The boards
are not bad from them, but I honestly dont trust the China fabs.
[image: image.png]

MAybe if you email them and give them this order number Order Number:
Y5-2327302A you can get them to make them. I ordered 5 boards and it cost
me $360, so its still not cheap with the shipping.

shred, 2021-03-16:
Thank you for your help! Yes, it's not cheap. As I don't know much about PCB making, it was too risky for me to get five nice wall decorations because I made a mistake somewhere. I found a reseller of your PCB, so I just ordered one in the meantime.

However it's good that it is now documented somewhere.

closed 2021-03-16 by shred